CodethinkLabs/mt3620-m4-drivers #3: GPIO's ConfigurePin() also needs to reset opposite configuration

dgarrett, 2020-03-05:
Currently ConfigurePin() in gpio.c sets the mask to oe_set if output or ie_set if input, but this results in a conflicting configuration if a pin is changed from output to input or vice versa.

ConfigurePin needs to first set the mask to oe_reset before setting ie_set, and ie_reset before setting oe_set to avoid a conflicting input/output configuration.

dgarrett, 2020-03-06:
It might be better even to just leave ie alone (defaults to enabled) and only set/unset oe. It looks like messing with ie might conflict with how the Linux driver handles GPIOs for HighLevel apps.

aiden-ct, 2020-03-06:
Thanks @dgarrett! Fixed in 3ed3666e. Re: "It might be better even to just leave ie alone (defaults to enabled) and only set/unset oe" - `ies` is input enable (not interrupt enable).

closed 2020-03-06 by aiden-ct