CodethinkLabs/mt3620-m4-drivers #6: ADC driver hangs on ADC_ReadSync

dgarrett, 2020-03-07:
I'm trying to do a one-shot read of ADC using ADC_ReadSync, but it just hangs waiting for the interrupt handler, which never fires.

I have a simple test app. In main it calls `VectorTableInit();` and `CPUFreq_Set(26000000);`

```
    AdcContext *handle = ADC_Open(MT3620_UNIT_ADC0);
    if (!handle) {
            return false;
    }

#define ADC_MAX_VAL 0xFFF
#define ADC_DATA_SIZE 8
    uint32_t rawData[ADC_DATA_SIZE];
    ADC_Data data[ADC_DATA_SIZE];
    int res = ADC_ReadSync(handle, ADC_DATA_SIZE, data, rawData, 0xF, 2500);
    if (res != ERROR_NONE) {
        ADC_Close(handle);
        return false;
    }

    for (int i = 0; i < ADC_DATA_SIZE; ++i) {
        uint32_t mV = (data[i].value * 2500) / ADC_MAX_VAL;
        WriteBuf(log, "ADC sample: mV ", mV);

        const int expectedVoltage = 1300;
        if (abs(mV - expectedVoltage) > 100)
        {
            ADC_Close(handle);
            return false;
        }
    }

    ADC_Close(handle);
```

dgarrett, 2020-03-08:
I figured it out. One issue was that rawData needs to be in SYSRAM, but it still hangs after that.

The other issue was that rawData was big enough to fit 8 measurements, but I passed in 0xF to ADC_ReadSync for the channel parameter. The ADC interrupt never fires because it only reads 4 channels which never fills the buffer, so it never triggers the buffer full interrupt.

I was able to fix this in my code by changing 0xF to 0xFF since I actually want all 8 channels, but the code should probably check for this case with ADC_ReadSync.

aiden-ct, 2020-03-09:
Good spot. Added porotection in 9e213b3eca98af5fe3772feff3a2f099b4814c48.

closed 2020-03-09 by aiden-ct