CodethinkLabs/mt3620-m4-drivers #7: GPIO driver doesn't support all GPIOs properly

dgarrett, 2020-03-10:
The GPIO driver only works correctly with GPIOs 0-23 and 41-48 (ADC block).

The other GPIOs that are associated with ISU blocks and I2S blocks don't work.

Can you please add support for GPIOs in ISU blocks and I2S blocks?

aiden-ct, 2020-03-11:
@dgarrett - I've added support in the `aiden/extend-gpio` branch. The tests I added to the samples repo indicate something's not quite right. Will debug tomorrow, but have added my WIP to same name branches on samples and drivers, so you can test yourself.

aiden-ct, 2020-03-11:
Note that to run the test with the Seed dev board, I've added some extra instructions to the README.md.

aiden-ct, 2020-03-13:
Fixed in ebd74150ff60bbccd26cc394fe70e141a907b95a.

closed 2020-03-13 by aiden-ct