CodethinkLabs/mt3620-m4-drivers #8: Is I2S_Input() function good?

xiongyu0523, 2020-03-12:
Hi, Codethink

I'm trying to use I2S driver to capture audio data from a analog microphone connect a DA7212 codec. I basically have codec configured and it do generate some data on RX line and I can measure by a logic analyzer.  At the same I'm streaming some test data to TX line to monitor if both direction of i2s driver works well.

1) I can see the exact the same data on TX line I write in output callback. I assume TX is working well
2) In input callback, I get the same data that I had written to TX buffer, this is not correct. I expect I can capture the same data as I see on the RX line. Not sure what goes wrong. 

My question is whether this driver is tested with both direction? Because I only saw the streaming use case in example..

My code is [here](https://github.com/xiongyu0523/azure-sphere-rtcore-da7212).. I do not have a single byte change to the driver, most of code are a port of da2712 i2c driver and it should affect what we see on i2s line and registers..

aiden-ct, 2020-03-13:
In your callbacks you're assuming size is in samples, but it's actually in bytes. We should document this better in the header.

The input buffer actually comes directly after the output buffer in memory, so I'd guess what's happening is that when you write the TX bytes, you're fully overwriting the RX buffer.

I'll update the documentation to make this more clear, thanks for reporting and let us know how it goes.

xiongyu0523, 2020-03-14:
Yes. I was aware that size is in bytes but I incorrectly apply byte index to a uint16_t array. It is quite error prone since data is passed using uint16_t pointer but size is in bytes..

xiongyu0523, 2020-03-14:
A follow up questions about I2S driver. 

The design is using callback to fill output data and receive input data. And it works continuously. What should I do if I have no data to be transmitted or just do not want (ready) to receive input data? Simply return false seems not good, I notice if I just return false in input callback, the output callback will not be called soon. In driver source code, the interrupt just return if user return false in callback, I guess there is no process in place to stop and reset that direction?

Previously I use NXP SAI driver and ST I2S driver. Both are using API to actively trigger a transaction and provide a buffer to transmit or to receive data. The callback  is just to notify the action is finished. User may be able to queue more than one transaction in the middle for ping-pong operation so audio will never drop. This seems more easily program model for developer.. Is it possible to change to this kind of driver? Thanks a lot for support

ben-brewer-codethink, 2020-03-16:
Just a quick note, the callback provides a void* rather than uint16_t* for this reason, it's only in the sample that we use uint16_t* and that's because we know that we set the sample size to 16-bits.

aiden-ct, 2020-03-16:
@xiongyu0523 - you should be able to pause output by calling:
```
I2S_Output(handle, I2S_FORMAT_NONE, 0, 0, 0, NULL);
```
You could use this to add a `Pause` method to the API that saves and recovers state. We are accepting PRs, so please feel free to contribute!

closed 2020-03-16 by aiden-ct

aiden-ct, 2020-03-23:
@xiongyu0523 - would you be able / happy to add your sample to our samples repo?