CodethinkLabs/mt3620-m4-drivers #17: Add a void * parameters to async complete callback?

xiongyu0523, 2020-10-11:
The current async API callback does not have a void* parameter that user can leverage to pass a data structure pointer. This is often required when user need to some thing in callback, e.g. post a semaphore or store a result. Also, current implementation is using a global variable, e.g. `I2CMaster_TransferSequentialSync_Status` to receive the status of all transfers. There is risk when two threads call the 'sync' transfer API and get incorrect status returned.

Prototype should be

```
int32_t I2CMaster_TransferSequentialAsync(
    I2CMaster *handle, uint16_t address,
    const I2C_Transfer *transfer, uint32_t count,
    void (*callback)(int32_t status, uintptr_t count));
```

-->

```
int32_t I2CMaster_TransferSequentialAsync(
    I2CMaster *handle, uint16_t address,
    const I2C_Transfer *transfer, uint32_t count,
    void (*callback)(int32_t status, uintptr_t count, void *user_data),
    void *user_data);
```

I can submit a PR if you're fine with this change.

aiden-ct, 2020-10-12:
Hi @xiongyu0523 - yes I think this is a good idea. I've just implemented similar for the MBox drivers. I'll have a look at some of the other drivers to see where this makes sense adding also.

xiongyu0523, 2020-10-12:
Great, let me know if you have plan. I will integrate with my project so I can submodule the whole driver library repo. thanks