CodethinkLabs/mt3620-m4-drivers #21: MT3620_I2C_QUEUE_DEPTH is 3, but the parameters check only allows up to 2 queued transfer

xiongyu0523, 2020-10-19:
Is it better to set MT3620_I2C_QUEUE_DEPTH to 2 and check using > instead of >= ?

ben-brewer-codethink, 2020-10-19:
Looks like it might be a bug, and since we never queued more than 2 transactions we never noticed. The hardware certainly claims to support 3 transactions queued, though I'm not sure why and have never tested to confirm it actually works.

ben-brewer-codethink, 2020-10-19:
Are you using a transaction queue of size 3 for anything?

xiongyu0523, 2020-10-19:
No. I don't.  I was thinking two transactions is good enough and also in MTK MHAL, it is [limited to two transactions](https://github.com/MediaTek-Labs/mt3620_m4_software/blob/f4adb3528f59733c5b4b224c6723fb5e42f463b8/MT3620_M4_Driver/MHAL/src/mhal_i2c.c#L702) in WRRD use case.

aiden-ct, 2020-12-09:
This has been tested and the HW does in fact support 3 transactions. Fix applied in PR#23.

closed 2020-12-09 by aiden-ct