CodethinkLabs/mt3620-m4-drivers #28: UART with Hardware Flow Control

breaultjean, 2022-03-02:
Hi,

It seems the UART driver as of today does not support the RTS/CTS hardware flow out of the box. We would like to _set_ the RTS pin when there is data in the TX send buffer and ensure it is _unset_ when there is no data in it. The idea is to interface with a RS485 transceiver in a non duplex scenario, similar to: 
[image]

Is there a way to enable this scenario to your knowledge?

**What I tried so far:**
- Control the pin by using the GPIO driver, did not work.
- Set the auto_rts flag in the Uart_Open function.
[image]

**More info on environment:**
- Azure Sphere SDK: 21.10.2.52856
- Sphere is the AES-MS-MT3620-M00-G
- Lib is at the latest from master (commit 0c4c772)

ben-brewer-codethink, 2022-04-28:
This should be fixed in the latest commit.

closed 2022-04-28 by ben-brewer-codethink

breaultjean, 2022-04-28:
Great! Thanks a lot.